ProjectNeura/LEADS #35: A Kinetic Energy Recovery System

ATATC, 2024-01-31:
This feature request is reserved for VeC 3.0.

qmascarenhas, 2024-02-04:
Easiest implementation would be to employ hub motors (which would also facilitate easier LEADS control) and then look at the different drive-regen options.

ATATC, 2024-05-30:
@qmascarenhas What battery are we using?

qmascarenhas, 2024-05-30:
For the 12 volt race, [Interstate MTX-35](https://www.interstatebatteries.com/products/mtx-35)
For the 24 volt race, [Optima Yellowtop SC35DA](https://www.optimabatteries.com/products/yellowtop-d35)

ATATC, 2024-05-31:
This enhancement will start once Thor has been revealed.

ATATC, 2024-05-31:
@qmascarenhas Is it required for the 24v race to connect the two batteries in parallel?

qmascarenhas, 2024-06-01:
2 12V batteries in series gets you 24V.