ProjectNeura/LEADS #37: The Bounce Time of the Wheel Speed Sensors

ATATC, 2024-02-01:
Assume the maximum speed is 260 km/h = 72.2222 m/s, and the wheel radius is 19 inches. The diameter is 96.25 cm = 0.9625m. The number of turns per second is 75.0361. The minimum time it takes to complete each full rotation is 13.3269 ms. Therefore, the bounce time of the wheel speed sensors is now set to 13 ms.

ATATC, 2024-02-01:
OK, this is not working properly at all. It is being triggered multiple times during one round. We need a new algorithm.

ATATC, 2024-02-01:
Bringing angular acceleration into  consideration may be the solution. By limiting the angular acceleration we might be able to dynamically adjust the bounce time.

qmascarenhas, 2024-02-01:
I haven't had a chance to look at the code but maybe we can discuss this before the 3pm team meeting today.

ATATC, 2024-02-01:
Another possible solution is to add an acceleration sensor as a reference.

qmascarenhas, 2024-02-01:
Yes, might not be a bad idea to add an accelerometer ... could be useful for other purposes.

ATATC, 2024-02-02:
Bounce time turns out to be an ineffective way to pop up the noise.
If the input is plotted in relation to time, it looks like this:
```
.....          .....         ....        ...       ..     ..
```

qmascarenhas, 2024-02-02:
1. Does the 5 dot sequence at the left represent the peak of the magnet entry and exit?
2. Is this pattern repeatable (or somewhat repeatable) for each entry and exit of the magnet?

ATATC, 2024-02-03:
The pattern shows that every time the magnet is close enough to the sensor, the sensor will be consecutively triggered for a period of time, and the period ends as it departs farther enough. The sensor reads at a variable rates, yet, in the triggering period, every time it reads it is triggered. We recognize this pattern as the entry and the exit, which means we have a bounce time that is equivalent to the sensor's detection interval.

ATATC, 2024-02-04:
> The pattern shows that every time the magnet is close enough to the sensor, the sensor will be consecutively triggered for a period of time, and the period ends as it departs farther enough. The sensor reads at a variable rate, yet, in the triggering period, every time it reads it is triggered. We recognize this pattern as the entry and the exit, which means we have a bounce time that is equivalent to the sensor's detection interval. If the wheel is running fast enough that it is meant to trigger the sensor constantly, the algorithm will fail to validate that input and interpret it as 0 RPM.

This has been proved effective.

qmascarenhas, 2024-02-04:
Can run an experiment on different diameter wheels to pin down a linear distance during which the sensor is triggered while in the field of the magnet? Our target wheel is 20". We would want to find the optimal distance away from the axle to mount the magnet and specify a mechanical mount for the sensor.

ATATC, 2024-02-04:
It's about less than 2 cm by visual inspection.

ATATC, 2024-02-04:
It looks just fine now except for two drawbacks.
1. The feedback rate depends on the speed of the wheel
2. Cannot determine zero speed

ATATC, 2024-02-04:
Btw I've completed the adapter that provides the connection between the Raspberry Pi (I'm using a laptop instead for debugging purposes, but it should work just the same on the Raspberry Pi) and the Arduino.

ATATC, 2024-02-09:
Zero speed determination algorithm will be discussed in a separate issue.

closed 2024-02-10 by ATATC