ProjectNeura/LEADS #39: LEADS BoM

qmascarenhas, 2024-02-03:
Bill of Materials for realizing LEADS

qmascarenhas, 2024-02-03:
@ATATC need to update the [spreadsheet]() I sent you please with details of everything you have purchased thus far.

ATATC, 2024-02-03:
Sure. I'll label this issue and do that sheet this weekend.

ATATC, 2024-02-06:
I have filled out the components purchased by now.

ATATC, 2024-02-08:
Ummm I don't think I have all the receipts.

qmascarenhas, 2024-02-08:
Those will work. Please post.

ATATC, 2024-02-09:
I'm gonna post it when the billing period comes.

qmascarenhas, 2024-02-22:
@ATATC Have you posted your CC transaction records for your expenses yet? Need that asap please.

ATATC, 2024-02-23:
[image]
[image]
[image]

ATATC, 2024-03-04:
# Component Proposal: GPS Receiver
@qmascarenhas Please review.

## GPS Breakout
https://www.sparkfun.com/products/17285

## Antenna
https://www.sparkfun.com/products/23814

## TTL to USB Converter
https://www.sparkfun.com/products/24639

## Qwiic Cable
https://www.sparkfun.com/products/14426

qmascarenhas, 2024-03-16:
> ![image](https://private-user-images.githubusercontent.com/52603299/307199791-e098d16a-f737-40b5-aa14-9154a5145378.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2MDA0NzMsIm5iZiI6MTcxMDYwMDE3MywicGF0aCI6Ii81MjYwMzI5OS8zMDcxOTk3OTEtZTA5OGQxNmEtZjczNy00MGI1LWFhMTQtOTE1NGE1MTQ1Mzc4LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxNlQxNDQyNTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYjY3NDRlZDM0ZjdkNTA2NjFiN2YxYjNlNDk5NmM5NmY5NWQ0OTU5NGE4MzM3ZjY4OGIzMjU1NzIzZWM0ZTA2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.eeA-3_DrVMRzz7v9Ns00-vjEpzKhJph8aLSQhju25v4) ![image](https://private-user-images.githubusercontent.com/52603299/307199798-4a760931-c04b-413a-9dd5-9a605eebbe5d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2MDA0NzMsIm5iZiI6MTcxMDYwMDE3MywicGF0aCI6Ii81MjYwMzI5OS8zMDcxOTk3OTgtNGE3NjA5MzEtYzA0Yi00MTNhLTlkZDUtOWE2MDVlZWJiZTVkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxNlQxNDQyNTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYzMwZmFjMDJjMmIzNzc0YzQyYWIyNTQwMTZiZjg5MzdmODliYTA4NWMyMTM0MWUzMTJiNzJkZmFkM2U5Mjk3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.CEOhtEw2ssaarU-aElmvh65zhbVlk2wZHGvdaN3V9V0) ![image](https://private-user-images.githubusercontent.com/52603299/307199813-a746241a-8ecb-4c79-b395-b2d1f8bd8d65.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2MDA0NzMsIm5iZiI6MTcxMDYwMDE3MywicGF0aCI6Ii81MjYwMzI5OS8zMDcxOTk4MTMtYTc0NjI0MWEtOGVjYi00Yzc5LWIzOTUtYjJkMWY4YmQ4ZDY1LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxNlQxNDQyNTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMTUwZDg3ZDhhYjEwNWQ1MmJjN2ZlNmVhYzIyNTBlZjA1ZmZmNGRjZGZmZWJiMGY5OWYwNTE1YzhlYjAxMzhmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bd_gMS8y8JNflEnJp5P_8kDJjMBsFG7jyLZGIH4dbws)

@ATATC does your online cc statement have a transaction details that you can print out? There should be a 

> # Component Proposal: GPS Receiver
> @qmascarenhas Please review.
> 
> ## GPS Breakout
> https://www.sparkfun.com/products/17285
> 
> ## Antenna
> https://www.sparkfun.com/products/23814
> 
> ## TTL to USB Converter
> https://www.sparkfun.com/products/24639
> 
> ## Qwiic Cable
> https://www.sparkfun.com/products/14426

This would work but product is out of stock.

@ATATC Would it make any sense for us to get one of the guys to work on writing an app to get our GPS co-ordinates from the phone? If we're having to use a phone for our internet connection, it would make sense for us to use some of the built-in features of the phone. Might be easier if our phone was an android, but I don't think that would be much of an issue. We should try to write an app that is independent of iPhone or Android and port accordingly.

ATATC, 2024-03-16:
Using a phone to record GPS coordinates is a feasible idea, but I have to point out some significant advantages and obstacles in advance so that we don't waste our time on it if we don't have someone capable.

I have been leading a project to write a native Android App called ArRow. Native apps are not able to run on multiple platforms. iOS uses Swift or Objective C and Andoird uses Java or Kotlin. I found it very brutal to write a native Android App in pure Java. Of course, there are cross-platform frameworks such as [flutter](https://flutter.dev/) that simplify the development, but they are usually difficult for beginners to get started if they don't know the basics of iOS and Android.

Another issue is debugging. Breakpoints cannot be inserted if the app is running on a phone. It has to be a simulator. But iOS simulators are only available in XCode on Mac, which means the developers must have a macOS device.

ATATC, 2024-03-16:
By the way, using TCP connections can be also challenging. There are many bugs that you could not expect. In the project (ArRow) I mentioned, I tried to build a long TCP connection library that supports both Python and Java but we failed due to the surprising workload. See [ATCP](https://github.com/ATATCTech/ATCP). But fortunately, the RSA encryption library [ARSA](https://github.com/ATATCTech/ARSA) continues to the present.

qmascarenhas, 2024-03-16:
Ok scratch that idea for now then - I don't think we have that experience or skill level on our team and it would take too long to develop. We just need to find a suitable GPS sensor that will give us the raw data we're looking for as I don't think we can access gps data live from a server ex: export live gps data for an active race with Strava.

ATATC, 2024-03-16:
What do you mean by export live gps data for an active race with Strava?

qmascarenhas, 2024-03-17:
For the last couple of races we used the Strava app on the phone and logged in remotely via the web (on laptop in the pit) to have a real-time position of the vehicle on the track. But I don't think the app allows real-time download of gpx data. The gpx data is only available when the race is ended (tracking stops) on the phone.
There's quite a few phone gps tracking apps out there that kinda do what we are trying to do. [LocaToWeb](https://locatoweb.com) is one that we could use if push comes to shove.
I still think there's good application and learning value for us to be able to implement obtaining gps data based on our original idea - a usb based GPS sensor - suited for raspberry pi applications. The kit you found developed by sparkfun would work well I think - the only problem is that it's out of stock.

qmascarenhas, 2024-03-17:
Of course we could go with the [u.fl rf antenna connector](https://www.sparkfun.com/products/15712) - it's just that those connectors can be a bit problematic if we plug and imply the antenna. The SMA connector is better for our purposes but not a show stopper.

ATATC, 2024-03-17:
We can go with assembled USB GPS receivers on Amazon.

ATATC, 2024-03-23:
@qmascarenhas https://www.amazon.ca/Navigation-External-Receiver-Raspberry-Geekstory/dp/B078Y52FGQ/ref=sr_1_2_sspa?crid=2XXTFD7YO36G7&dib=eyJ2IjoiMSJ9.tilH0HvRvDYEwApcuenax2lgwBGzW3xm5bKo6Glet0fFP0U7Z8_Xi1cr3hHUCAfRzfuowJaPU3aUfw0D3vS8R7eU1H4eQipQ3tcDVWBJvpmaoKJr2PPV4EyaXYIHOxoQYdTNyKNfW4yqRllnyaoGKGcSyN1Tk3e32Edr71r4kbBAIkv9NHfoEbKUjh5zVpuMYfh5IV6rSploNK65qMNgfkayQE6CWLYYEAzR2wNMt6CcWRguIEsCO7vs2Zy0UFIQxVOWpFvFkFgdRMb_U4WkTu0jXcSdEEFecBVWobCZEqo.XBI5MlaVeTOvgrD-qypPfTT-tf0rLo73JaUK44jEXBE&dib_tag=se&keywords=usb%2Bgps%2Breceiver&qid=1711206728&sprefix=USB%2BGPS%2Caps%2C75&sr=8-2-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1

qmascarenhas, 2024-03-23:
> @qmascarenhas [...]

Added and ordered. @ATATC should have it for you on Monday!

ATATC, 2024-04-08:
@qmascarenhas Maybe we should work in this way: I'll cover the cost of the components I bought and they count as mine and then replace them with the ones purchased by the school so that I can finish it as my AP and computer science assignment faster and we won't run into billing issues.

ATATC, 2024-04-08:
I can reuse these components in the future as well so there won't be any waste.

qmascarenhas, 2024-04-09:
_> @qmascarenhas Maybe we should work in this way: I'll cover the cost of the components I bought and they count as mine and then replace them with the ones purchased by the school so that I can finish it as my AP and computer science assignment faster and we won't run into billing issues._

I'm ok with this plan but you should still log these in the LEADS BoM s/s.

qmascarenhas, 2024-04-10:
> ![image](https://private-user-images.githubusercontent.com/52603299/307199791-e098d16a-f737-40b5-aa14-9154a5145378.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI2NzAxNzksIm5iZiI6MTcxMjY2OTg3OSwicGF0aCI6Ii81MjYwMzI5OS8zMDcxOTk3OTEtZTA5OGQxNmEtZjczNy00MGI1LWFhMTQtOTE1NGE1MTQ1Mzc4LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwOVQxMzM3NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ODVhNjFmOGY0MjFhNGU0Y2VmZTJiM2ZjYWE1M2RiMTZlNDIxZWZlNWFjMmJmMjg5MTg2MDMwYzc4ZjFlOTY1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.SXO-8vRR8dEYPqinP6Vdb3bU9c4-K_qtv6jqjjlmPoE) ![image](https://private-user-images.githubusercontent.com/52603299/307199798-4a760931-c04b-413a-9dd5-9a605eebbe5d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI2NzAxNzksIm5iZiI6MTcxMjY2OTg3OSwicGF0aCI6Ii81MjYwMzI5OS8zMDcxOTk3OTgtNGE3NjA5MzEtYzA0Yi00MTNhLTlkZDUtOWE2MDVlZWJiZTVkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwOVQxMzM3NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYmNmNzM2ODUxZmM5ODMyZDQ5OGEwNGVlMmUyN2I1YjE3OWYyZDFlMmYzZmZlYWFiYWVmZGM3YjM5NmE0M2E5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.x6ty31Hhb30M9vJDF3Gn1o-w1hWRTizR2uDOZHop8KI) ![image](https://private-user-images.githubusercontent.com/52603299/307199813-a746241a-8ecb-4c79-b395-b2d1f8bd8d65.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI2NzAxNzksIm5iZiI6MTcxMjY2OTg3OSwicGF0aCI6Ii81MjYwMzI5OS8zMDcxOTk4MTMtYTc0NjI0MWEtOGVjYi00Yzc5LWIzOTUtYjJkMWY4YmQ4ZDY1LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwOVQxMzM3NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05NTRiZmE3NzE2MzFlNTczZGQyNjIxODk2OWJmMGNjNGU4ZjcxYTc4YzdjN2I4ZmEwYWYwMDUwYWY3NTRiODA5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.d8bs6gl8-wlNWGhHuaTE4K-vTum1VeKvyv5lEUFajJg)

@ATATC After our conversation yesterday about your receipts I went back to the spreadsheet and created a [TFu Purchases Tab](https://villanovaonline.sharepoint.com/:x:/s/VeC2.0DesignTeam-InstrumentationandControl/EWdYYaZJ5oVGg0v6int1vs8BXSmzvYt4wo8G6wt23t6ACg?e=M3doiQ&nav=MTVfezAwMDAwMDAwLTAwMDEtMDAwMC0wMTAwLTAwMDAwMDAwMDAwMH0). Please review and edit as necessary and let me know when you have completed the edit. _**I am unable to reconcile your CC Transaction record amounts with the items purchased. I need to be able to reconcile these amounts in order to claim the expense.**_

ATATC, 2024-04-10:
OK. I'll do it when I have time.

ATATC, 2024-06-12:
Now, this issue is moved to email.

closed 2024-06-12 by ATATC