ProjectNeura/LEADS #40: Expose LEADS VeC RC Backend to the Cloud

ATATC, 2024-02-04:
As the car may not be under the same local network as the Pit crew, we may need to use Frp to expose the middle layer to the cloud.

ATATC, 2024-04-13:
We will run frpc on the Raspberry Pi which connects to the frps on the cloud server. A laptop connects to the Raspberry Pi through the cloud server. The cloud server's job is also known as reversed proxy. Finally, the frontend website connects to the laptop (localhost or shared NAT).

Learn more about frp: https://github.com/fatedier/frp

@qmascarenhas

qmascarenhas, 2024-04-13:
@ATATC I think you need to update the diagram in your docs to indicate the FRP service option for the remote analyst if we use a phone hotspot in the vehicle:
[image]
For LEADS as tested presently, the "vehicle" and the remote analyst are on the same network.

ATATC, 2024-04-13:
The diagram is correct. It's just that LEADS VeC RC and the front end are integrated into one icon which is the laptop labeled as "Pit Crew".
This is the cloud server:
[image]

ATATC, 2024-04-13:
I did not note "frp" in the diagram because the middleware may differ.

qmascarenhas, 2024-04-13:
@ATATC can you cost out how much it would cost for us to have a temporary FRP cloud server for 1-2 months (April-May)?

ATATC, 2024-04-13:
If we use [Lightsail](https://lightsail.aws.amazon.com/ls/webapp/home/instances), a service by AWS, we have a 3-month free trial.

ATATC, 2024-04-13:
Otherwise, it starts from 3.5 USD/month.

ATATC, 2024-04-13:
I would recommend AWS as that's what my cloud data center uses.

qmascarenhas, 2024-04-13:
> If we use [Lightsail](https://lightsail.aws.amazon.com/ls/webapp/home/instances), a service by AWS, we have a 3-month free trial.

Ok, let's go ahead with this then and add as a ToDo to cancel after the race.

ATATC, 2024-04-13:
Can you sign up for a new account? I don't have that offer.

qmascarenhas, 2024-04-13:
> The diagram is correct. It's just that LEADS VeC RC and the front end are integrated into one icon which is the laptop labeled as "Pit Crew". This is the cloud server: [...]

Maybe for clarity on the diagram we should include a *NOTE for "Pit Crew" that indicates that Pit Crew includes:

1. LEADS VeC RC & F/E
2. MS-Teams Video Call with LEADS vehicle

and another **NOTE on the Cloud indicating some middleware like FRP will be required unless the "Pit Crew" and LEADS vehicle are on the same LAN.

qmascarenhas, 2024-04-13:
> Can you sign up for a new account? I don't have that offer.

I've gone ahead and done this but you're going to have to lead me through setup and/or creating a user account for you that will work for what you need to do. Probably best if you do this when we are in the same room.

ATATC, 2024-04-17:
Figured out the issue. We should create the user in AWS IAM Identity Center instead of IAM.

ATATC, 2024-05-03:
We have built the reverse proxy system on https://leads-proxy.projectneura.org. Configurations will not be uploaded to the repository due to security considerations, but we have it tested during the open house.