ProjectNeura/LEADS #83: Speedometer

SAKajiura, 2024-03-28:
I think that we should use analog speedometer because it is easier to read.

Maybe we could create something that looks similar to the image below.

[image]

ATATC, 2024-03-28:
Are you suggesting a new hardware speedometer or an enhancement to the software?

ATATC, 2024-03-28:
Default font sizes have been increased. An update for the preview screenshot will be scheduled.

ATATC, 2024-03-29:
[image]

ATATC, 2024-03-29:
You can still change it to the classic style by clicking it.

ATATC, 2024-03-29:
NEXT
- [ ] Rename "mode" to "style"
- [ ] Support specifying style
- [ ] Support disabling default click call back that switches the style

SAKajiura, 2024-03-29:
Looks good.

ATATC, 2024-03-30:
NEXT

- [ ] Implement the third style.

ATATC, 2024-03-30:
[image]
[image]
[image]

closed 2024-03-30 by ATATC

ATATC, 2024-03-31:
# Improved UI
[image]
[image]

The color of the arc changes according to the speed.