ProjectNeura/LEADS #86: Implement a Hardware Low Pass Filter

ATATC, 2024-03-29:
[image]

# Function
## Input
A PWM analog input
## Output
The average voltage of the PWM input

qmascarenhas, 2024-03-29:
@ATATC can you point me to the codeset instruction where you are generating the input to the LPF? Are you sure it is a [PWM](https://www.electronics-tutorials.ws/blog/pulse-width-modulation.html) signal? The signal you are showing on the diagram is not a PWM signal.
Anyway, assuming it is a PWM signal, a [single pole RC (passive) LPF should suffice](https://www.allaboutcircuits.com/technical-articles/low-pass-filter-a-pwm-signal-into-an-analog-voltage/).

ATATC, 2024-03-29:
That blog seems to be what we want. I did not really look into the diagram.

This is how we control the motor:
https://gpiozero.readthedocs.io/en/stable/api_output.html#motor

closed 2024-08-01 by ATATC