ProjectNeura/LEADS #87: Implement the G Force Recording System

ATATC, 2024-04-01:
- Complete `GForceMeter`
- Include G force data in the default data container yet
- Create data persistence for G force data in `leads_vec_rc` 
- Include G force data in the remote analyst

qmascarenhas, 2024-04-01:
Does the rpi4 have a built-in accelerometer?

ATATC, 2024-04-01:
Unfortunately no.

ATATC, 2024-04-01:
We need an external at least 6 DoF accelerometer.

qmascarenhas, 2024-04-01:
here's where the Phyphox feature would come in handy.

ATATC, 2024-04-03:
Bug found: render delay significantly increases after 30 seconds.

ATATC, 2024-04-03:
Resources are wasted on unnecessary re-rendering.

ATATC, 2024-04-03:
Replacing the dynamically drawn circles with a fixed image is worth trying.

ATATC, 2024-04-03:
> CPU usage still increases extremely as long as the G force meter is rendered.

It is confirmed that this is caused only by the G force meter.

ATATC, 2024-04-04:
Investigating.

ATATC, 2024-04-04:
Bug fixed.

ATATC, 2024-09-09:
https://learn.adafruit.com/adafruit-9-dof-orientation-imu-fusion-breakout-bno085/uart-rvc-for-arduino

closed 2024-09-27 by ATATC