ProjectNeura/LEADS #91: Component Proposal: 9-DOF IMU (Accelerometer)

ATATC, 2024-04-05:
As it is described in #87, we need a hardware accelerometer.
The following module manufactured by Bosch is definitely outstanding as it provides a built-in fusion algorithm. Usually, these algorithms consume huge amounts of time to develop.
> Rather than spending weeks or months fiddling with algorithms of varying accuracy and complexity, you can have meaningful sensor data in minutes thanks to the BNO055
[image]
https://www.adafruit.com/product/4646?gad_source=1&gclid=CjwKCAjwwr6wBhBcEiwAfMEQs25ZsRQSI3ZZr5F1jnn5Rk_EU1TZqa_kDwB_7Oy19h_hhXwNvQKq0xoCyzcQAvD_BwE

ATATC, 2024-06-30:
## Correction
I found BNO085 is a better and cheaper option, just that it's out of stock on Adafruit. You'll have to buy it on DigiKey.
https://www.adafruit.com/product/4754

ATATC, 2024-06-30:
@qmascarenhas Please still review this proposal as the acceleration data is going to play a role in this year's development.

ATATC, 2024-08-06:
@qmascarenhas Please still review this proposal as the acceleration data is going to play a role in this year's development

qmascarenhas, 2024-08-07:
@ATATC ordered.

ATATC, 2024-09-09:
Soldering required.

ATATC, 2024-09-27:
The other side also needs soldering. P0 needs to be pulled as high to specify UART mode.

closed 2024-09-28 by ATATC