ProjectNeura/LEADS #92: LEADS First Assembly

ATATC, 2024-04-06:
@ProjectNeura/vec-instrumentation 

As the development of LEADS reaches a certain stage, it is time to put our results into practice. Before entering the formal assembly, we first need to complete the following adjustments.
- The header on the Arduino should be removed because we would like to wire multiple cables onto one pin
- A second Arduino Micro should be purchased
- The measurements of the components should be done

Afterward, the assembly will take place hopefully in the following order.
1. Wheel speed sensors onboard and tested
2. Wheel speed subsystem Arduino onboard and tested
3. Raspberry Pi onboard and tested
4. Display onboard and tested
5. GPS onboard and tested
6. Voltage sensor onboard and tested
7. Power subsystem Arduino onboard and tested

ATATC, 2024-04-06:
Please confirm a date to do the soldering work.

qmascarenhas, 2024-04-06:
Wednesday April 10 after school? This can really be done any day after school once I show you what you need to do (assuming you have not soldered before).

qmascarenhas, 2024-04-06:
Would it make sense for us to get a [passive heat sink protective case for the rpi4](https://www.pishop.ca/product/armor-case-passive-cooling-for-raspberry-pi-4-model-b-black/)?

ATATC, 2024-04-06:
It already has a small heat sink. We can design an air deflector though.

ATATC, 2024-04-06:
I don't think the screen case is necessary because the control buttons are behind the screen. But for aesthetic reasons, we can design a border to hide the PCB.

ATATC, 2024-04-06:
> Tuesday April 9 after school? This can really be done any day after school once I show you what you need to do (assuming you have not soldered before).

Sure. I have soldered before. I just don't have the equipment and site here.

ATATC, 2024-04-06:
By the way I figured out that the touchscreen needs the USB power cable to connect to the raspberry Pi. We definitely need some USB hubs.

qmascarenhas, 2024-04-07:
Since the cameras and the OBS-Teams call is separate from the main LEADS s/w (I'm assuming we'll be able to launch all that with a script instead of manually), we can put the cameras on a hub which would leave us 3 usb ports on the rpi4:
1 - Touch Display
2 - GPS dongle
3 - open

ATATC, 2024-04-07:
Actually the keyboard and mouse still take 2 ports.

ATATC, 2024-04-07:
Here's what we need to connect to the USB ports. Each listing takes one port.
- 3 cameras
- GPS receiver
- Wheel speed controller
- Power controller
- Display power
- Display touchscreen
- Keyboard
- Mouse

ATATC, 2024-04-07:
We need 4x 3-port USB hub.

ATATC, 2024-04-07:
Do we have enough USB hubs? If not I can go get some tmr.

qmascarenhas, 2024-04-07:
We've got lots of hubs at school that are not being used that we should be able to use. Please don't buy anything.

ATATC, 2024-04-11:
When will we be ready to install the system on VeC99?

qmascarenhas, 2024-04-11:
I'm hoping we will have VeC99 on the road by the end of next week.
We need to put an assembly plan together for your system first in terms of final layout on VeC99. Can you put a draft plan together that I can review?

1. Target the area under the hood.
2. Layout components other than the display, powerpack and sensors on a panel that will fit on the underside of the hood at an angle. We will cut some vents in the top of the front panel of the nose to direct some air flow under the hood for cooling.

ps: your usb hubs are in (I've put them with the rest of your stuff under the nose in rm217. I have not tested them.

- [Anker 4-Port USB 3.0 Ultra Slim Data Hub](https://www.amazon.ca/dp/B00XMD7KPU?ref=ppx_yo2ov_dt_b_product_details&th=1)
- [Aceele 4-Port USB 3.0 Ultra Slim Data Hub w USB-C Power Port](https://www.amazon.ca/dp/B07RM53SZK?ref=ppx_yo2ov_dt_b_product_details&th=1)

ATATC, 2024-04-12:
The new USB hubs have been tested and the system worked well.

ATATC, 2024-04-23:
@qmascarenhas When do you think the car is going to be ready so that we can film a video demonstrating the system.

qmascarenhas, 2024-05-03:
@ATATC @fz20181223luogu We need to test the MS-Teams call with the OBS virtual cam combining the images from Cameras 1, 2 and 3.

- @fz20181223luogu you need to provide @ATATC with your OBS virtual cam configuration.

qmascarenhas, 2024-05-09:
@ATATC can you bring the HDMI cable that I gave you in tomorrow? I'll swap it for the one that's on VeC99 - it's just better for cable routing under the hood and is a straight cable without need of adapters.

ATATC, 2024-05-10:
- [ ] Reinstall the operating system after tests

ATATC, 2024-05-10:
@ProjectNeura/vec-instrumentation After tons of bug fixes now finally the program can start itself when the Raspberry Pi boots up.

ATATC, 2024-05-12:
@qmascarenhas What do you think about giving @fz20181223luogu a chance to touch the core of LEADS, that to let him follow the [user guide](https://leads-docs.projectneura.org/en/latest/vec/user-manual.html) to reinstall the operating system for the other Raspberry Pi?

qmascarenhas, 2024-05-12:
I concur with this plan @ATATC - I would also like to do something similar with @HaydenHour but like you he writes the AP Physics 1 exam on Friday so perhaps after that. I will recommend him to get his own raspberry pi so he can play around with it over the summer.

ATATC, 2024-05-21:
## May 21
- [ ] Install OBS on the Raspberry Pi B
- [ ] Fix #179
- [ ] Test with longer cables
- [ ] Complete #139

ATATC, 2024-05-22:
@qmascarenhas Sir, when you do the installation today can you leave a secondary HDMI port and cable for quick debugging purposes?

qmascarenhas, 2024-05-22:
Not quite sure what you mean by _"leave a secondary HDMI port and cable"_.

ATATC, 2024-05-22:
I mean plug another HDMI cable into the Raspberry Pi and put the other end somewhere accessible without dismantling the car.

qmascarenhas, 2024-05-22:
ok. I'll need to get another cable though - I only bought one the last time.

ATATC, 2024-05-22:
Sure. I'm not at school today but keep me in the loop with the installation progress.

ATATC, 2024-05-22:
@qmascarenhas Any update today?

qmascarenhas, 2024-05-23:
I ran out of time to put everything together. I made the cables for the wheel speed sensors. I'm going to try getting in early tomorrow so I can get some work done before P1.

ATATC, 2024-05-23:
I can do some work during P3. I'll finish the system installation on the original Raspberry Pi that we'll use on the car and finish testing the voltage sensor. But we are still struggling to put everything under the hood if the car is used for practice tomorrow after school, aren't we?

qmascarenhas, 2024-05-23:
The plan is have everything prepped for installation on VeC99 after they bring the vehicle back to the shop from practice. We should be able to pre-test everything on the cart (not ideal but it is what it is) and do our final test on the vehicle after installation. It's going to be a late one tonight.

ATATC, 2024-05-25:
@qmascarenhas I've included an alternative option for us in the new release.

qmascarenhas, 2024-05-30:
A week behind schedule, but LEADS finally installed on VeC99 for full system testing.[image][image][image][image][image][image][image]

ATATC, 2024-05-31:
@qmascarenhas Sir, did you connect the wheel speed sensors according to the [User Guide](https://leads-docs.projectneura.org/en/latest/vec/user-manual.html#connect-the-wheel-speed-sensors)?

qmascarenhas, 2024-05-31:
No - probably not. I just figured you had the data pins on the arduino dictated in a config file. Sorry - forgot to check could this be the cause of our erroneous speed readings? At least one wheel should be reading the correct speed if that's the case.

closed 2024-06-02 by ATATC

qmascarenhas, 2024-06-02:
@ATATC I'm going to do some further testing of LEADS on VeC99 after exams and before my summer break starts. Can you spare some time after your last exam to run me through the setup of the camera feeds for the livestream? That's the only thing pending for us to test live. I'm going to create another issue titled "Detailed Testing of LEADS Analog Sensors" that I will own for now and then assign it one of our junior members after I detail a test plan. If there's one thing we need to spend more time on before releasing LEADS on VeC99 to the racing team it would be a full functional test. We can title that issue "Release of LEADS on VeC99 to VeC Racing".

ATATC, 2024-06-03:
My last exam is at 6.12 in the afternoon (due to the conflict between Bio and Math). I'm free that day afterward. I'll also finish the documentation for GStreamer.

ATATC, 2024-06-03:
Sorry, in fact, the documentation is complete: https://leads-docs.projectneura.org/en/latest/vec/user-manual.html#install-gstreamer-optional