ProjectNeura/LEADS #103: Redesign the Home Page

ATATC, 2024-04-12:
During the past versions, leads.projectneura.org has always been the documentation page. Now we are considering building a single-slide project homepage. This is an open task that anyone can contribute to. The website will be a separate repository (not created yet).

This task involves
- Website design
- Frontend tech stack
- HTML
- CSS
- Javascript
- Version Control through GitHub

ATATC, 2024-04-12:
@qmascarenhas Maybe choose someone to do this as it is one of the easier ones?

qmascarenhas, 2024-04-13:
@fz20181223luogu Would you be interested in taking this on?

ATATC, 2024-04-19:
An example:
[image]

fz20181223luogu, 2024-04-19:
> @fz20181223luogu Would you be interested in taking this on?

I'm afraid that I could not, since I have very little knowledge about programming languages related to websites.

ATATC, 2024-04-19:
These languages are very easy to learn. If you could do this it would save us lots of time.

ATATC, 2024-05-03:
@fz20181223luogu Any update?

closed 2024-05-22 by ATATC