ProjectNeura/LEADS #130: OBS Studio Fails to Initialize

ATATC, 2024-05-06:
In today's test, OBS Studio failed to run on Ubuntu 22.04 LTS for Raspberry Pi with the following error message:
"OBS failed to initialize video. Your GPU is not supported or your graphics drivers may need to be updated."

ATATC, 2024-05-06:
@fz20181223luogu @qmascarenhas I was not able to reproduce this issue on the virtual machine. I'm afraid that we have to schedule another test.

qmascarenhas, 2024-05-06:
@ATATC we await your heads up for when we can do this based on your AP Exam schedule. @fz20181223luogu, while we wait for Terry, we can test Teams on the rpi with one of the cameras. This will also allow us to familiarize ourselves with the boot-up procedures for LEADS. Terry, if you can direct us to any updated docs on this, that would be great!

ATATC, 2024-05-06:
Running Teams is extremely simple just like it is on Windows. I'll send you the password of the Raspberry Pi on Edsby.

ATATC, 2024-05-07:
I'm trying to run an arm64 version of Ubuntu 22.04 LTS virtual machine on my Mac.

qmascarenhas, 2024-05-08:
> Running Teams is extremely simple just like it is on Windows. I'll send you the password of the Raspberry Pi on Edsby.

@ATATC Can you send me the password? I assume the rpi boots up on pwr connection.

qmascarenhas, 2024-05-08:
> @ATATC we await your heads up for when we can do this based on your AP Exam schedule. @fz20181223luogu, while we wait for Terry, we can test Teams on the rpi with one of the cameras. This will also [...]

@fz20181223luogu [Completed test today](https://villanovaonline.sharepoint.com/sites/VeC2.0DesignTeam-InstrumentationandControl/Shared%20Documents/Recordings/Meeting%20in%20_Instrumentation%20(and%20Control)_-20240508_151303-Meeting%20Recording.mp4?web=1&referrer=Teams.TEAMS-ELECTRON&referrerScenario=MeetingChicletGetLink.view.view) - MS-Teams runs fine and can switch between the 3 cameras on on the hub. We now need to figure out what the issue with OBS on the rpi.

ATATC, 2024-05-08:
I believe you would probably paste that in #85.

ATATC, 2024-05-09:
According to https://forums.raspberrypi.com/viewtopic.php?t=337544, looks like exporting `LIBGL_ALWAYS_SOFTWARE=true` could fix this issue. However, this env var means that it always use no hardware acceleration. It could possibly lead to low performance.

ATATC, 2024-05-09:
Since I do not have the Raspberry Pi, I'll put a possible solution here.
1. Update the System
```shell
apt update
apt full-upgrade
```
2. Configure the Driver
```shell
raspi-config
```
- Navigate to Advanced Options
- Select GL Driver
- Choose GL (Fake KMS)
- Exit the configuration tool and reboot the Raspberry Pi
3. Verify
```shell
apt install mesa-utils
glxinfo | grep "OpenGL version"
```

ATATC, 2024-05-09:
Another worth-trying solution is adding this to `/boot/config.txt`:
```txt
[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
```

ATATC, 2024-05-09:
## Possible Cause
After the last test on the OBS Studio in December, there has been a new version released. In version `30.1.2` [change log](https://launchpadlibrarian.net/723387594/obs-studio_30.1.2-0obsproject1~jammy_source.changes), it says "Note: NVIDIA users may need to update their GPU drivers to 531.61 or newer." I assume a similar deprecation happened to the Raspberry Pi.
However our Raspberry Pi 4 Model B 8GB only supports OpenGL ES 3.1 and no Vulkan. Support for Vulkan 1.2 starts from Raspberry Pi 5.

qmascarenhas, 2024-05-09:
Today after school, we'll try the updates you suggest first and if that doesn't resolve the problem, we might have to deprecate the OBS Studio version we are using. If we can't resolve the issue we'll just have to use a single camera.

ATATC, 2024-05-09:
We can always integrate camera view merging into LEADS. I proposed using OBS just to reduce the workload. Python can easily implement a virtual camera as well, but as always, without enough time spent on optimizing, the performance will always be worse than the existing solutions (I spent a week just to get rid of the redundant rerenders). If we really run out of fixes with OBS Studio I can spend an afternoon implementing the identical feature as a new package probably called `leads_svm` (surround view monitor) in Python, just that it has to be after May 17. It'll be used to replace OBS Studio, not Teams.

ATATC, 2024-05-09:
But anyway we will eventually get rid of OBS Studio for two reasons.
1. OBS Studio is not headless
2. We need camera inputs for deep learning based features

qmascarenhas, 2024-05-09:
Ok. That makes sense - hopefully we can get OBS working easily but if not, we have some options that would still give us a video feed for Teams.

ATATC, 2024-05-09:
Just note that the Python virtual camera relies on vfl2loopback. The capability has not been verified yet.

closed 2024-05-09 by ATATC