ProjectNeura/LEADS #139: Teams for Linux is Out of Date

ATATC, 2024-05-09:
Since Teams for Linux is not officially supported, we might want to consider migrating to other streaming platforms such as Discord or Zoom (these are officially released), or even better, use YouTube directly so that we don't have to run any other app.
@qmascarenhas @fz20181223luogu How does it sound?

qmascarenhas, 2024-05-09:
Streaming directly to YouTube (unlisted) would actually be better for us. I'm trying to get @fz20181223luogu to use OBS Studio to produce a Livestream production of the event with OBS Studio. The YouTube stream would be one of his feeds.
ps: @ATATC can you bring the HDMI cable that I gave you in tomorrow? I'll swap it for the one that's on VeC99 - it's just better for cable routing under the hood and is a straight cable without need of adapters.

ATATC, 2024-05-09:
For YouTube, we just need to get a streaming key and put that into the OBS Studio and it'll just work hopefully (lots of things don't work like they should on Raspberry Pi).

qmascarenhas, 2024-05-09:
Worth a try!

ATATC, 2024-05-19:
@qmascarenhas @fz20181223luogu Have you tested this proposal?

qmascarenhas, 2024-05-20:
I was going to try the YouTube livestream with OBS today but just realized that it takes 24-h to get that enabled.

ATATC, 2024-05-24:
Succeeded in #193.

closed 2024-05-24 by ATATC