ProjectNeura/LEADS #171: Test: Remote Analyst Stress Test with Reverse Proxy

ATATC, 2024-05-14:
@qmascarenhas We should schedule a stress test for the whole data chain of the remote analyst and the DNS, CDN, and caching middlewares along with reverse proxy.

qmascarenhas, 2024-05-14:
Yes, absolutely! We could do this as early as this Friday after your AP Physics 1 exam if you are sticking around.

ATATC, 2024-05-16:
Yes sure. But I really want to test the system on the actual car because of the GPS.

qmascarenhas, 2024-05-16:
I do too - I'll be working on doing integrating everything on the under-hood shelf tomorrow while you are writing your AP Physics 1 exam. They are planning on some race testing tomorrow after school so if I can get things done before that, we can have both at the same time although I would prefer to do our testing separately. Are you free at all this long w/e during the day?

ATATC, 2024-05-16:
Yes. I can stay till 5 if we need that much time.

ATATC, 2024-05-17:
Today's data:
[2024-05-17_17-08-03.csv](https://github.com/ProjectNeura/LEADS/files/15358502/2024-05-17_17-08-03.csv)

closed 2024-05-17 by ATATC