ProjectNeura/LEADS #173: Add Postprocess Analysis

ATATC, 2024-05-14:
As it is discussed in the meeting, we want some post-process analysis as it leaves us more time to run more complex algorithms.

ATATC, 2024-05-16:
There are still some errors in identifying laps.

ATATC, 2024-05-16:
## Demo
@qmascarenhas Check this out.

A lap from last year's data:

[image]

Comparison between all laps last year:

[image]

ATATC, 2024-05-16:
- [ ] Add a label for the color bar

ATATC, 2024-05-16:
[image]

qmascarenhas, 2024-05-16:
@ATATC That's awesome!!

ATATC, 2024-05-17:
- [ ] Add suggestion generators

ATATC, 2024-05-17:
- [ ] Add a page on the docs website to generate scripts

ATATC, 2024-05-18:
- [ ] Make `matplotlib` optiona
- [ ] Export the static methods to a separate file
- [ ] Export the inferences to a separate file

closed 2024-05-19 by ATATC