ProjectNeura/LEADS #178: Improve the Tolerance of the Wheel Speed Algorithm

ATATC, 2024-05-17:
As it was noted during today's test #171, the wheel speed sensor frequently failed to read a value due to the shaking of the wheels. Therefore, we want to improve the tolerance of the algorithm so that it does not compromise easily.

Also, it has been brought to attention that although the vehicle was at rest, the program still outputs some random speeds. Further investigation is required.

ATATC, 2024-05-18:
## A Multi-sensor Hypothetical Fusion Inference Algorithm

When there is a **huge** difference between the readings of different wheel speed sensors, we assert that there is failure. We use other available valid data to infer the speed. If multiple inference suggest speeds that fit in an acceptable range, take the smallest reading the lies in that range as the final output.

Definition of "**huge**": the difference is greater than the overall variance of the speed sequence.

### Implementation
The implementation of this algorithm deeply depends on the utilities introduced in #173. I'm going to make the inferences available without `matplotlib`.
Little bit concerned about the performance, but should be fine.

@qmascarenhas Please review.

ATATC, 2024-05-19:
@qmascarenhas This is way too hard and time-consuming to realize. I won't be able to finish it within a week. I propose we go with better mounting methods.

qmascarenhas, 2024-05-19:
No problem @ATATC - it shouldn't be too difficult to find a way to reliably secure the magnet to the wheel spoke without it twisting due to wheel/spoke vibration. We can postpone this task to after the race.

ATATC, 2024-05-19:
In my investigation of the zero speed problem (replaying mode really helped), I suspect that it might be caused by the "enhancement" of the wheel speed algorithm introduced 4 months earlier. I'll try to fix that first. I suggest we schedule another test next week before the race.

qmascarenhas, 2024-05-19:
We can test all we want on the temporary cart setup we have. We need to assign priority to the test drives this week to get the drivers and pit crew race ready. Our work on and with VeC99 is going to have to happen outside of their time with the vehicle until we can assure them that we have a stable solution that will not disrupt their practice routine.
We need to be able to race VeC99 without LEADS so if there is a LEADS catastrophic failure on race day there is no panic.

closed 2024-05-19 by ATATC

ATATC, 2024-05-19:
Do we still have enough time to install the instrumentation system?

qmascarenhas, 2024-05-19:
Yes! I need to do one more test on the cart with the longer wheel speed sensor wires (those measurements you took on Friday) and we should be good. If you could give me a hand with that this week while I iron out some of the mechanical installation issues, that would be great!

ATATC, 2024-05-19:
Yes sure. I should be available any afternoon this week.

ATATC, 2024-05-20:
Wednesday is off. Other days are free.

qmascarenhas, 2024-05-20:
ok. Let's try to work to the following schedule then:

- Tue May 21 3pm - 4:30pm: Test LEADS on cart with the longer wires for the 3 wheel speed sensors and live video stream from OBS to YouTube.
- Wed May 22: _I will be working on the LEADS mechanicals for integration under the hood of VeC99_
- Thu May 23: 3pm - 5pm: Final testing of LEADS on VeC99 in simulated racing situation.

ATATC, 2024-05-20:
Do we have a time schedule for the race day?

qmascarenhas, 2024-05-20:
![https://uwaterloo.ca/electric-vehicle-challenge/races/waterloo-ev-challenge]([image](https://github.com/ProjectNeura/LEADS/assets/151644374/77c7bd7b-0c4b-4f86-afc5-153529cfdec0)
)

ATATC, 2024-05-20:
Can't see it.

qmascarenhas, 2024-05-20:
i just updated the comment.

ATATC, 2024-05-20:
Did u tell Jerry that he needs to come to my place at 7?

qmascarenhas, 2024-05-21:
No I didn't. Can you sort that out with him directly?