ProjectNeura/LEADS #180: Test: Final Testing of LEADS on VeC99 in Simulated Racing Situation

ATATC, 2024-05-21:
See https://github.com/ProjectNeura/LEADS/issues/178#issuecomment-2121061196.

ATATC, 2024-05-21:
- [ ] Voltage sensor

ATATC, 2024-05-21:
@qmascarenhas I can do some tests on Thursday during period 3.

ATATC, 2024-05-22:
If the voltage reading is still off, we can simply use the current hardware voltage sensor.

ATATC, 2024-05-24:
We ran out of time today, will continue tomorrow.

ATATC, 2024-05-25:
Ran out of time.

closed 2024-05-25 by ATATC

qmascarenhas, 2024-05-26:
[Rescheduled LEADS Final Testing: Wednesday May 29 14:30-17:30 VC lower parking lot.](You're invited to LEADS Final Testing on VeC99
Wednesday, May 29, 2024
16:30 - 17:30 (EST)
https://teams.microsoft.com/l/meetup-join/19%3ajQixv-o19Ib_eAQkUmCRv45l6UBOUd-yCcSydYUl_5g1%40thread.tacv2/1716755580132?context=%7b%22Tid%22%3a%22fa41f09f-512c-4efa-ab1a-3f288fe37a3c%22%2c%22Oid%22%3a%22b50c8c4b-520f-4197-90ee-2b61490f8b8a%22%7d
Tap on the link or paste it in a browser to join.)

ATATC, 2024-05-30:
[2024-05-30_16-15-41.csv](https://github.com/ProjectNeura/LEADS/files/15505513/2024-05-30_16-15-41.csv)

ATATC, 2024-05-31:
## Data
[2024-05-30_16-15-41.csv](https://github.com/ProjectNeura/LEADS/files/15505513/2024-05-30_16-15-41.csv)
## Conclusion
In today's test, the system was working perfectly except for some issues encountered.
The wheel speeds was not reading stably correctly. It can be because of the improper mounting mechanism from #92.
The voltage reading was jumping in a considerably wide range. It seems to be the inaccuracy of the sensor.

qmascarenhas, 2024-05-31:
@ATATC we should state those comments as "Observations" and in "Conclusion" propose some testing to confirm hypothesis of the issues.
- I will upload shortly the .gpx file from Strava for comparison with LEADS. At first glance (map plot) correlation seems to be tight.
- The voltage during drive will fluctuate as we load the battery. We need to have some comparison between LEADS and a reliable terminal voltage sensing. I have NI myDAQ that we can use for this.
- The wheel speed sensor zone of optimal sensing needs to be determined properly and the test we did comparing LEADS and Cateye Velo needs to be repeated. After moving the magnet on the rear wheel to optimize sensing for LEADS we lost sensing on the Cateye Velo so we don't have any data to correlate to other than gps.

qmascarenhas, 2024-05-31:
Strava Data (.gpx) to correlate with [LEADS (gps) data](https://github.com/ProjectNeura/LEADS/files/15505513/2024-05-30_16-15-41.csv): 
[LEADS_on_VeC99_Test_1.zip](https://github.com/user-attachments/files/15513705/LEADS_on_VeC99_Test_1.zip)