ProjectNeura/LEADS #182: Error Voltage Readings

ATATC, 2024-05-21:
We found in #92 that the voltage sensor readings were off.

ATATC, 2024-05-21:
Temporarily fixed.

qmascarenhas, 2024-05-22:
@ATATC can you point me to the code for this?

ATATC, 2024-05-22:
If you find on the page,
[image]
Click there and you will be redirected to the change.

ATATC, 2024-05-22:
And the Arduino code: https://github.com/ProjectNeura/LEADS-Arduino/commit/c2d573d26187338d900b5aaef2aa08859ae94189

ATATC, 2024-05-22:
@qmascarenhas Pointed out a more reasonable solution (https://github.com/ProjectNeura/LEADS-Arduino/commit/c2d573d26187338d900b5aaef2aa08859ae94189#commitcomment-142288646).

ATATC, 2024-05-22:
I've updated the algorithm according to the resistance stated here: https://www.osepp.com/downloads/pdf/VoltageSensor.pdf.

closed 2024-05-23 by ATATC