ProjectNeura/LEADS #206: Support Pedals as Built-in Devices

ATATC, 2024-06-02:
As we've discussed, over the summer, @HaydenHour is assigned to add support for pedals.

qmascarenhas, 2024-06-04:
@HaydenHour see #207 for test plan details of analog sensors - it should provide a framework for functional testing of your summer task. We can discuss further at our meeting on Monday.

ATATC, 2024-06-26:
@HaydenHour Hi Hayden, how it is going?

HaydenHour, 2024-06-28:
@ATATC Hi Terry, its going all right so far. I've decided to do a couple small projects to acclimate myself to using C. In a bit I am planning to do some stuff with the micro Arduino, and eventually beginning to complete the support for pedals as we discussed

HaydenHour, 2024-07-26:
@ATATC I have mostly completed the initial pedal. The video is blurry, but a value between 0-1 is being outputted based on the throttles position. I just want to ticker a bit with the code to make it a bit better. As expected, there are a couple variables in the code that are dependent on certain calibrations (i.e. the initial non 0 rest value being returned).

https://github.com/user-attachments/assets/0743f43b-9bf5-4157-8721-1311dfbb102a

ATATC, 2024-07-26:
Great work.

Now you may want to wrap the code in LEADS framework as I described in the meeting earlier.

As an example, this is the voltage sensor's code:
```cpp
#ifndef VOLTAGESENSOR_H
#define VOLTAGESENSOR_H

#include "Device.h"

class VoltageSensor : public Device<float> {
protected:
    float _factor;
public:
    VoltageSensor(float r1, float r2, int *const pins);
    void initialize();
    float read();
};

#endif //VOLTAGESENSOR_H
```

``` cpp
#include "VoltageSensor.h"

VoltageSensor::VoltageSensor(float r1, float r2, int *const pins) : _factor((r1 + r2) / r2), Device<float>(pins) {}

void VoltageSensor::initialize() { pinMode(_pins[0], INPUT); }

float VoltageSensor::read() { return (float) analogRead(_pins[0]) * _factor * 5 / 1023; }
```

You can download the library here: https://leads-docs.projectneura.org/en/latest/arduino/index.html.

In this example, the dependent variables are `r1` and `r2` in the constructor.

ATATC, 2024-07-28:
Sorry for the confusion, we are recently preparing for a massive launch of breaking changes to our Arduino Library. A corresponding revision will be released then. The newer version shall prevail. You may wait until then.

HaydenHour, 2024-08-03:
@ATATC Hi Terry, sorry I haven't responded sooner, I have been a bit busy with family. Just to avoid confusion, should I reconfigure my code to resemble what you have shown, or is this something that should be done in a separate file?

ATATC, 2024-08-04:
@HaydenHour
Basically, your project structure should be like this:
```
LEADS
-arduino
--leads_vec_power # open your Arduino IDE here
---leads_vec_power.ino
---Pedal.h
---Pedal.cpp
```

ATATC, 2024-12-02:
@qmascarenhas @HaydenHour how are we doing on this?

closed 2025-03-04 by ATATC