ProjectNeura/LEADS #207: Detailed Testing of LEADS Analog Sensors

qmascarenhas, 2024-06-02:
For full functionality, LEADS requires a number of analog sensors. The first official release of LEADS for VeC99 Racing requires:

1. A Voltage Sensor - _to indicate BATT voltage (and eventually infer and indicate the BATT capacity once we have collected enough data. At the moment there is no indicated inference of BAT capacity)._
2. 3 Wheel speed Sensors - _for each of the 2 20" front wheels and the single 26" rear drive wheel._

For a future release, the following analog sensors would also be useful:

1. Throttle position Sensor - _to indicate %age of depression (0 to max) of throttle lever or pedal._
2. Controller Error Code Indicator - _to indicate the error code number based on LED blinks for controllers (like the current KDS24200E deployed on VeC99). More expensive controllers will facilitate reading of error codes via a CAN/SPI or I^2^C interface._

The rpi mini-computer does not support direct reading of analog signals - only digital via the GPIO bus - because it does not have a built-in DAC. To overcome this, the initial release of LEADS uses several (2) Arduino micros (one for each sensor class) to input a digital value between 0 and 1023 that represents a voltage level serially to the rpi via a USB interface.

**Testing Protocol for Analog Sensors Deployed on LEADS**
**BATT Voltage Sensor**
This sensor consists of a simple [30kOhm/7.5kOhm voltage divider](https://www.amazon.ca/DIYmall-Voltage-Sensor-Arduino-DC0-25V/dp/B00NK4L97Q) for a BATT Voltage range of 0 to 25V. The 10-bit DAC on the Arduino micro converts this input level to a 10-bit binary number that is input digitally to the rpi via the USB interface. To validate the voltage sensed, a calibrated secondary source like a digital multimeter or a [NI myDAQ](https://www.ni.com/docs/en-US/bundle/mydaq-specs/page/specs.html) (for easy data logging of voltage sensed) should be used.

**Wheel Speed Sensor**
This sensor consists of a [Hall Switch](https://www.pishop.ca/product/analog-hall-effect-magnetic-sensor-module-arduino-compatible/) which indicates a high or a low depending on whether or not the sensor is in the vicinity of a magnetic field (of a magnet mounted on a spinning wheel). This on/off state will indicate the period of one revolution of the wheel (if a single magnet on the wheel is used). From this information and the size of the wheel, wheel speed is determined and input digitally to the rpi via the USB interface. To validate the wheel speed sensed, a calibrated secondary source like the [Cateye Velo wireless](https://www.amazon.ca/Cateye-Velo-Wireless-Speedometer-2017/dp/B0183WLS52/ref=asc_df_B0183WLS52/?tag=googleshopc0c-20&linkCode=df0&hvadid=335176345855&hvpos=&hvnetw=g&hvrand=296326400858344447&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9000960&hvtargid=pla-308082964977&psc=1&mcid=6786db51f72f3290acafa0e183d0de82) should be used for ease of testing different wheels.

ATATC, 2024-06-22:
@qmascarenhas I bought a new type of hall switch that is smaller, cheaper, and more sensitive.

qmascarenhas, 2024-06-22:
@ATATC please [see](https://github.com/ProjectNeura/Thor/issues/2#issuecomment-2184037881) - _I am unable to read any details of the components you are purchasing!_