ProjectNeura/LEADS #254: Test: OpenCV Performance on Raspberry Pi

ATATC, 2024-07-02:
Huge progress has been made in regard to video support in the past few pull requests. However, concerns about the performance of the Raspberry Pi do exist as the testing platform, Apply M3 Pro, is only able to provide a video feed at almost 30fps. Therefore, it is likely that OpenCV runs as slowly as GStreamer does on the Raspberry Pi. This poor performance wouldn't affect the dashboard anyway though.

Ideally, this is what we have accomplished so far:
[image]
[image]

The monitor is just an additional widget. You can still switch back to other widgets.
It now supports up to 4 cameras, of which 1 is enabled in this case, which are front view, left view, right view, and rear view respectively.

ATATC, 2024-08-07:
The performance of the raspberry barely supports OpenCV. The bottleneck seems to be encoding and decoding.

ATATC, 2024-08-08:
Trying on a virtual machine to see the minimum requirement.

ATATC, 2024-08-08:
The bottleneck is Pillow.

ATATC, 2024-08-08:
If use OpenCV instead, it consumes 10%.

ATATC, 2024-09-06:
@qmascarenhas Is it possible to schedule this test for Monday during the second period or after school?

qmascarenhas, 2024-09-07:
yes - we could do that.

ATATC, 2024-09-08:
@qmascarenhas That period just got occupied. What about Tuesday's last period?

ATATC, 2024-09-10:
Raspberry Pi fails to run `devices_visual.py`.

closed 2024-09-10 by ATATC