ProjectNeura/LEADS #464: Support National Instruments myDAQ as an Input Source

ATATC, 2025-01-31:
As @qmascarenhas requested, I have investigated the Python binding for National Instruments myDAQ and believe that it can be used as an authentic source for both sensor calibration and experiments.

ATATC, 2025-01-31:
Reference: https://github.com/ni/nidaqmx-python

ATATC, 2025-02-21:
@qmascarenhas Can you lend me a sensor of any type to test the myDAQ tomorrow?

qmascarenhas, 2025-02-27:
> [@qmascarenhas](https://github.com/qmascarenhas) Can you lend me a sensor of any type to test the myDAQ tomorrow?

yes, sure! sorry I missed this ... can you lmk what exactly you need?

ATATC, 2025-02-27:
What sensors do we have?

qmascarenhas, 2025-02-27:
For sensing, the myDAQ has analog and digital inputs. I would focus right now on the analog inputs see [example1_read_AI_pins.py](https://github.com/daskals/NI-myDAQ-Python-examples/blob/main/example1_read_AI_pins.py). You could even use the myDAQ itself ... wire the 5V pwr output to the AI channel, read and log the reading. I could also give you one of the throttles and you can wire it up to the 5V pwr output and measure the throttle output (AI channel - read and log). In fact, this could be something that @HaydenHour should do as an exercise for his work on the [E&M Lab Series work for Lab 1](https://teams.microsoft.com/l/message/19:8VtmhDpdGgyl3Lo8kc0s4Fb8ga4_1mwTVrOaS1aNlcU1@thread.tacv2/1740673409265?tenantId=fa41f09f-512c-4efa-ab1a-3f288fe37a3c&groupId=8d9d27b5-3d40-4c7f-9f98-58fa4725e572&parentMessageId=1736540758319&teamName=SPH4Up.2024-25&channelName=General&createdTime=1740673409265).

ATATC, 2025-02-28:
Can you let @HaydenHour make a pull request to merge his previous work on the pedal first?

qmascarenhas, 2025-03-03:
@ATATC @HaydenHour you will need to install the NI-DAQmx driver before you try testing. See [https://github.com/ni/nidaqmx-python/blob/master/README.rst#getting-started](https://github.com/ni/nidaqmx-python/blob/master/README.rst#getting-started)

ATATC, 2025-03-03:
Btw the read and write functionalities for input and output devices are not yet implemented.

qmascarenhas, 2025-03-03:
> @ATATC @HaydenHour you will need to install the NI-DAQmx driver before you try testing. See [https://github.com/ni/nidaqmx-python/blob/master/README.rst#getting-started](https://github.com/ni/nidaqmx-python/blob/master/README.rst#getting-started)

@HaydenHour for testing I would suggest using VS instead of the wrapper that @ATATC has started writing just to make sure we have a handle on how to address the myDAQ. To start, I would wire DGND to AI0- and +5V to AI0+ and then follow the example steps to read the AI0 channel. You should expect to read a float value of around 5. Please drop by rm217 tomorrow to pick up some jumper wires that you can use.

closed 2025-05-27 by ATATC