ProjectNeura/LEADS #468: VeC 2.0

ATATC, 2025-02-20:
## VeC 2.0

@fz20181223luogu is now in charge of LEADS' integration in VeC 2.0. Yesterday, the new Orange Pi was distributed. Please actively update your progress.

### TODOs

- [ ] 1. Install the operating system on the Orange Pi
- [ ] 2. Install and correctly configure LEADS
- [ ] 3. Install and correctly configure FRP reverse proxy
- [ ] 4. Successfully connect the Remote Analyst Dashboard with the Orange Pi
- [ ] 5. Eliminate all fault lights

### Helpful Resources

- [User Manual](https://leads-docs.projectneura.org/en/latest/vec/user-manual.html) for steps 1 to 3
- [Remote Analyst Dashboard](leads-vec-rc.projectneura.org) for step 4

fz20181223luogu, 2025-03-26:
Now, the Ubuntu 24.04.1 LTS Orange Pi has been successfully installed on the Orange Pi, and the LEADS has been installed and correctly configured on it. The FRP reverse proxy has also been installed.

ATATC, 2025-03-26:
Good. Now please continue to step 4.

fz20181223luogu, 2025-03-27:
Now the Remote Analyst Dashboard has been successfully connected to the Orange Pi via the proxy server.

ATATC, 2025-03-27:
Post a screenshot.

fz20181223luogu, 2025-03-27:
> Post a screenshot.

[image]

ATATC, 2025-03-27:
Record a video showing that clicking the buttons on the Remote Analyst results in changes on the Orange Pi.

fz20181223luogu, 2025-03-27:
> Record a video showing that clicking the buttons on the Remote Analyst results in changes on the Orange Pi.

The video is uploaded to the Teams under files of Instrumentation channel

ATATC, 2025-03-28:
Saw it. Great job.

ATATC, 2025-03-28:
Ask @qmascarenhas for duplications of the hardware components of LEADS on VeC99, which include the Arduinos.

qmascarenhas, 2025-03-28:
> Ask [@qmascarenhas](https://github.com/qmascarenhas) for duplications of the hardware components of LEADS on VeC99, which include the Arduinos.

@fz20181223luogu as @ATATC requested, the 2 Arduinos have been purchased and are available for use. You need to make a list of all other hardware components that need to be duplicated including cabling please.

fz20181223luogu, 2025-03-28:
> [@fz20181223luogu](https://github.com/fz20181223luogu) as [@ATATC](https://github.com/ATATC) requested, the 2 Arduinos have been purchased and are available for use. You need to make a list of all [...]

I will take an investigation this weekend to list components needed.