ProjectNeura/LEADS #483: Wheel Speed Controller Harness

ATATC, 2025-03-05:
3 3-pin connectors with buckles.

DanielS928, 2025-03-06:
[image]

ATATC, 2025-03-06:
Where are we planning to install the fuses, led, resistor, and shutoff? Are they embedded into the headers?

Btw @qmascarenhas please review this schematic.

qmascarenhas, 2025-03-06:
> Where are we planning to install the fuses, led, resistor, and shutoff? Are they embedded into the headers?
> 
> Btw [@qmascarenhas](https://github.com/qmascarenhas) please review this schematic.

@ATATC I did review the schematic with @DanielS928. Please note that this particular schematic is for VeC99. The schematic for VeC 2.0 will depend on what configuration RDU we are using.

For VeC99, 

- fuses and resistor are on the rear electrical panel behind the driver and above the motor
- led is on the right front dash panel with the controller power switch
- emergency shut-off (main shut-off) is behind the left shoulder of the driver (accessible to both the driver and service crew in case of emergency)

DanielS928, 2025-03-24:
@qmascarenhas These are the connectors for the wheel speed sensors:
Male end: https://www.digikey.ca/en/products/detail/phoenix-contact/1829400/2525207
Female end: https://www.digikey.ca/en/products/detail/phoenix-contact/1803646/260536

ATATC, 2025-03-28:
> [@qmascarenhas](https://github.com/qmascarenhas) These are the connectors for the wheel speed sensors: Male end: https://www.digikey.ca/en/products/detail/phoenix-contact/1829400/2525207 Female end: https://www.digikey.ca/en/products/detail/phoenix-contact/1803646/260536

Do you want him to purchase it?

ATATC, 2025-05-02:
> https://shop.sayal.com/products/gmte-1451-1?_pos=54&_sid=1345a2be8&_ss=r

@qmascarenhas